ftdi-rs/ftdi-embedded-hal #16: I2C clock stretching support

kuon, 2022-06-12:
I use some STM32 micro controllers as slave and I need the I2C clock stretching functionality.

Is it possible to have it supported by this library with a FT232H?

newAM, 2022-06-12:
I do not think it will work with the MPSSE.  It might be possible if you bitbang with the pins setup as ordinary GPIOs, unfortunately this will result in a low clock speed since the GPIO interface is not aa fast.

Do you know of other FTDI libraries that support this?

kuon, 2022-06-12:
Apparently it is not supported directly, but pyftdi did it:

https://github.com/eblot/pyftdi/blob/b366832c5371b3b8849c3247d23116240c1b203f/pyftdi/doc/api/i2c.rst#clock-streching

Do you think we could implement something like this?