ftdi-rs/ftdi-embedded-hal #53: hal::FtHal::init_freq(device, 400_000)? expected `Error`, found `TimeoutError`

circuit4u-medium, 2024-05-27:
after
`let device = libftd2xx::Ft232h::with_description("USB <-> Serial Converter")?;`
then
`let hal = hal::FtHal::init_freq(device, 400_000)?;`

got this error message

[image]

newAM, 2024-05-27:
Can you share the full code, and which version of this crate you are using?

This is similar to code that gets tested in lib.rs, it should work.

circuit4u-medium, 2024-05-27:
Thanks for the quick reply. Yes, `cargo run` the code in examples folder doesn't have any issue. 

I was trying to run Rust code in a Jupyter notebook interactively. Here are the snippets:

[image]

newAM, 2024-05-27:
Can you try `:dep ftdi-embbeded-hal = { version = "0.22.0", features = ["libftd2xx", "libftd2xx-static"] }`?

I'm not sure how the resolver works in jupyter, it might not have enabled those.

circuit4u-medium, 2024-05-28:
Thanks for the advice. I think the issue is resolved, but now there seems to be a new one: 
[image]

newAM, 2024-05-28:
That's from the underlying libftd2xx driver.  Never seen that one before myself.  libftd2xx is closed source; you can debug as far as the driver call, which is probably worthwhile to check the handle (pointer) is reasonable.  There's also logging in the rust wrapper for the driver which will log the pointer.

If you're on Linux then the open source drivers are worth a try (requires different feature flags).  The open source drivers are easier to debug.