ftdi-rs/ftdi-embedded-hal #57: Implementing spi traits on the struct as opposed to a reference to the struct

alexkl-rugged-controls, 2024-10-09:
Lets say I have a struct which is supposed to store a generic `SpiDevice`:
```rust
struct Foo<E> {
    device: Box<dyn SpiDevice<Error = E>>
}
```
This works with `linux_embedded_hal` but not with `ftdi_embedded_hal`, because it appears the `SpiDevice` trait is only implemented on references to `SpiDevice<Device>`.

I'm fairly new to rust, but it seems that simply implementing it for the struct directly works for both references and owned values:
```rust
impl<Device, E> eh1::spi::SpiDevice for &'a SpiDevice<Device>
// into
impl<Device, E> eh1::spi::SpiDevice for SpiDevice<Device>
```
Is there a reason why it's only implemented for references?

newAM, 2025-03-09:
Sorry for the slow response, this got lost at the bottom of my inbox.

I think there was a reason for this in the past, but that no longer applies.

Thanks for opening the issue!

closed 2025-03-09 by newAM

newAM, 2025-03-09:
Fixed in release [v0.23.0](https://github.com/ftdi-rs/ftdi-embedded-hal/releases/tag/v0.23.0)

PR #62 (2025-03-09): SpiDevice: implement traits on struct instead of ref
Fixes #57