ftdi-rs/ftdi-embedded-hal #66: Read FTDI EEPROM from FtHal?

shareefj, 2025-03-24:
Hi there,

Sorry for the noob questions but I'm new to Rust.  So I've managed to get as far as using `FtHal` to create an `i2c` interface and successfully reading/writing the bus.  I'm now looking at being able to dump/program the FTDI EEPROM contents but can't work out the correct method to access this data.

I've modelled my physical board with the following type:

```
pub struct Board {
    hal: FtHal<Ft4232h>,
    i2c_interface: Option<ftdi_embedded_hal::I2c<Ft4232h>>,
    reset: OutputPin<Ft4232h>,
    en: OutputPin<Ft4232h>,
}
```

There doesn't seem to be any way to access the `eeprom_read` and `eeprom_program` methods from `FtHal`.  I think I need to call these on the underlying device but once you've created the `FtHal` object you're a bit stuck.

Any advice on the correct way to access the EEPROM traits?

newAM, 2025-03-25:
This crate isn't really meant for accessing the EEPROM.  The EEPROM has a few differences in APIs between libftd2xx and libftdi, and then also between each device, which makes it harder to abstract than the SPI, GPIO, and I2C busses.

You could access the EEPROM before turning it into an `FtHal`, but that may not be practical in all situations.

I could add a function that accepts a closure to read EEPROM things (or really do anything) on a concrete inner device, but I'm not sure if that can fully address your problem.  Something like this:

```rs
/// Executes the closure with the device.
///
/// Useful for accessing EEPROM, or other device-specific functionality.
///
/// # Example
///
/// ```no_run
/// use ftdi_embedded_hal as hal;
/// use hal::libftd2xx::FtdiEeprom;
///
/// # #[cfg(feature = "libftd2xx")]
/// # {
/// let device = libftd2xx::Ft2232h::with_description("Dual RS232-HS A")?;
/// let mut hal = hal::FtHal::init_default(device)?;
/// let serial_number: String =
///     hal.with_device(|d| d.eeprom_read().map(|(_, strings)| strings.serial_number()))?;
/// # }
/// # Ok::<(), std::boxed::Box<dyn std::error::Error>>(())
/// ```
pub fn with_device<T, F>(&mut self, mut f: F) -> T
where
    F: FnMut(&mut Device) -> T,
{
    let mut inner = self.mtx.lock().expect("Failed to aquire FTDI mutex");
    let result: T = f(&mut inner.ft);
    result
}
```

I opened #67 with that code.  If it works for you I'm happy to merge it!

closed 2025-03-28 by shareefj

newAM, 2025-03-29:
Fixed in v0.23.1